MrYsLab/s2aio #1: Problem finding configuration.cfg file on Mac Computers

MrYsLab, 2015-11-14:
The problem manifests itself in an exception that looks something like this:
```
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.5/bin/s2aio", line 9, in <module>
    load_entry_point('s2aio==1.1', 'console_scripts', 's2aio')()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/s2aio/__main__.py", line 806, in main
    s2aio = S2AIO(client=client_type, language=language_type, com_port=comport)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/s2aio/__main__.py", line 66, in __init__
    self.ln_languages = config.get('translation_lists', 'ln_languages').split(',')
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/configparser.py", line 779, in get
    d = self._unify_values(section, vars)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/configparser.py", line 1139, in _unify_values
    raise NoSectionError(section)
configparser.NoSectionError: No section: 'translation_lists'
```
This will be resolved ASAP

fisherds, 2015-11-15:
site-packages path on my Mac

/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/s2aio

MrYsLab, 2015-11-15:
Fixed with version 1.3. Changed using a static, OS specific set of python paths to a dynamically discovered path.

closed 2015-11-15 by MrYsLab