MrYsLab/s2aio #3: An exception occurred on the asyncio event loop

wesVelde, 2015-12-01:
Hello Alan,

I have been using s2a_fm for a while now and the kids love it. I wanted to try s2aio on my mac today however I get an error message.
I have done all the steps indicated, but get the following msg:

*pymata_aio Version 2.7	Copyright (c) 2015 Alan Yorinks All rights reserved.*
*Using COM Port:/dev/tty.usbmodem411*
*Initializing Arduino - Please wait... Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins*

*An exception occurred on the asyncio event loop while receiving data.  Invalid message.*

Can you help me?

Using mac os el capitan
Python 3.5

MrYsLab, 2015-12-01:
@wesVelde There was a change made to one of the packages (aiohttp) that s2aio uses. I recently updated the code to make it compatible with those changes, so you might want to try

pip install s2aio --upgrade

or perhaps:

 sudo pip s2aio --upgrade

or if you have pip3:

sudo pip3 s2aio --upgrade

This will make sure you have the latest and greatest version.

1. Did s2aio find the correct com port ? Is /dev/tty.usbmodem411 the port that the arduino is connected to? 

2. Do you have either StandardFirmata or FirmataPlus uploaded to the Arduino?

3. Try removing the USB cable form the computer and plugging it back in.

Please let me know if any of this fixes your problem.

MrYsLab, 2015-12-01:
An additional question if none of the above helps - if you start s2aio with the following option:

s2aio -c no_client

Do you still get the exception?

wesVelde, 2015-12-01:
Using s2aio -c no_client works. 

I opened the scratch file in /Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/s2aio/ScratchFiles/ScratchProjects and created a blink. Worked great!

Thank you for the quick responce.

closed 2015-12-01 by wesVelde