MrYsLab/s2aio #5: Failed to create process

JosepPablo, 2016-01-03:
Congratulations for your work,
I'm working with s2a_fm and I would try s2aio. I followed the steps on your wiki, for Windows S.O. in two different computers without problems (W7 and W10), but when I try to start with  the command "s2aio", I always get: " failed to create process." 
I'm a noob with Python and is possible that I missed some basic step, but I'm not solving this issue. 
Thanks a lot for any help.

MrYsLab, 2016-01-03:
Hi,

Is there an exception thrown or any other output other than " failed to create process" ?

Also, can you please try with the following command line and let me know if the same error occurs:

s2aio -c no_client

JosepPablo, 2016-01-04:
Thanks for your response,
I haven't any other output.
I attach the screen with the trials.

[image: Imatge inserida 2]

Josep

MrYsLab, 2016-01-04:
@JosepPablo,
Hi,
I don't think you can add an image to a comment. But if things are still not working let me ask a few more questions.

1. Do you have python3.5 installed? It is necessary in order to run s2aio - python2 will not work with it.
2. If you have both python 3.5 installed and python 2 installed, this is not a problem, but to do the following test, we need to determine how to start python 3. So if you type python from a command window, does it identify itself as python2  or python3. If it is python 2, type python3 or python3.5 to make sure that python 3.5 is installed.
3. Once you know how to start python3.5, we are going to try an manually start the program

I am assuming you installed into the default directories, so to start the program manually cd into:

C:\Users\YOUR USER NAME\AppData\Local\Programs\Python\Python35\Scripts

You should see this file in the directory: s2aio-script.py

Then start the program using whatever command you use to start python 3 (in the example below I am using python - but adjust for your system). Type the following:

python s2aio-script.py

Please let me know if you have any different results or if you any questions about the instructions above.

Thanks.

JosepPablo, 2016-01-05:
Hi Alan!

With your instructions I noticed that "python" was running the 2.7 version.

Now I'm trying: python3 s2aio-script.pywit the result:

*C:\Users\Josep Pablo\AppData\Local\Programs\Python\Python35-32\Scripts>p*
*2aio-script.py*

*pymata_aio Version 2.8  Copyright (c) 2015 Alan Yorinks All rights reser*

*Using COM Port:com3*

*Initializing Arduino - Please wait...*
*C:\Users\Josep Pablo\AppData\Local\Programs\Python\Python35-32\Scripts>*

But I don't know how to continue. I was waiting Scratch starts, but nothing
hapens.
Could be an incompatibility problem with s2a_fm installation?

Thanks!

Josep

MrYsLab, 2016-01-05:
Ok, so if Scratch has started, the first to check is to see if the blocks loaded properly and if the Arduino has connected to Scratch.

To do this, after Scratch opens, the "Motion" blocks are shown.  Click on the "More Blocks" tab in the blocks area of the editor. You should the s2aio blocks and there should be a connectivity indicator to the left of where is says s2aio. This should a green dot if there is connectivity or a red dot if there is not connectivity.

If you see the blocks and the green dot, you should be able to program your arduino.

Please let me know if things are working or not.

Thanks

JosepPablo, 2016-01-05:
I'm sorry, but Scratch never started.
I tried manually started but s2aio blocks didn't appear.
I tried with Snap but didn't started.

Thanks

Josep

MrYsLab, 2016-01-06:
Just to clarify things, you did say that you installed s2a_fm and that it works. Is that correct?

When I start s2aio, here is what I see on the console:

```
>s2aio

pymata_aio Version 2.8  Copyright (c) 2015 Alan Yorinks All rights reserved.

Using COM Port:com3

Initializing Arduino - Please wait... Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins
```

Is this what you see - that the number of pins has been identified and that the program continues to run until you kill it?

JosepPablo, 2016-01-07:
Yes, s2a_fm is working. I see in the console:

*c:\s2a_fm-master>c:\python27\python s2a_fm.py com3*
*s2a_fm version 1.5   Copyright(C) 2013-14 Alan Yorinks    All Rights
Reserved*

*Python Version 2.7.11 (v2.7.11:6d1b6a68f775, Dec  5 2015, 20:32:19) [MSC
v.1500*
*32 bit (Intel)]*

*PyMata version 2.11  Copyright(C) 2013-15 Alan Yorinks    All rights
reserved.*

*Opening Arduino Serial port com3*

*Please wait while Arduino is being detected. This can take up to 30
seconds ...*
*Board initialized in 0 seconds*
*Total Number of Pins Detected = 20*
*Total Number of Analog Pins Detected = 6*
*Please wait for Total Arduino Pin Discovery to complete. This can take up
to 30*
*additional seconds.*
*Arduino Total Pin Discovery completed in 0 seconds*
*Starting HTTP Server!*
*Use <Ctrl-C> to exit the extension*

*Please start Scratch or Snap!*

but when I try: python3 s2aio-scropt.py I see in my console:

*>python3 s2aio-script.py*

*pymata_aio Version 2.8  Copyright (c) 2015 Alan Yorinks All rights
reserved.*

*Using COM Port:com3*

*Initializing Arduino - Please wait...*
*>*

and the program stops before initializing Arduino.

Josep

MrYsLab, 2016-01-08:
To try and isolate the problem, could you please try the simple blink program [in this gist.](https://gist.github.com/MrYsLab/ba0964c4c76be6da4f28). Just cut and paste into a file on your computer and run it, and let me know if that works. What you should see is the L13 LED on the Arduino blink a few times. 

If this should fail as well, please change line 36 of the program from:
`
board = PymataCore()
`

to 
`
board = PymataCore(arduino_wait=5)
`
and try running again. This will increase the amount of time allowed for the Arduino to reset itself. Please let me know if this also fails.

Here is the output when I run the program on Windows using python3:

```
C:\Users\Alan>python new_blink.py

pymata_aio Version 2.8  Copyright (c) 2015 Alan Yorinks All rights reserved.

Using COM Port:com5

Initializing Arduino - Please wait... Auto-discovery complete. Found 30 Digital Pins and 12 Analog Pins

Shutting down ...

C:\Users\Alan>

```

JosepPablo, 2016-01-09:
Hi!

I tried it with the next output, after change the 36th line:

*C:\Users\Josep
Pablo\AppData\Local\Programs\Python\Python35-32\Scripts>python3 b*
*link.py*

*pymata_aio Version 2.8  Copyright (c) 2015 Alan Yorinks All rights
reserved.*

*Using COM Port:com3*

*Initializing Arduino - Please wait... Traceback (most recent call last):*
*  File "blink.py", line 41, in <module>*
*    loop.run_until_complete(board.start_aio())*

*  File "C:\Users\Josep
Pablo\AppData\Local\Programs\Python\Python35-32\lib\async*
*io\base_events.py", line 337, in run_until_complete*
*    return future.result()*

*  File "C:\Users\Josep
Pablo\AppData\Local\Programs\Python\Python35-32\lib\async*
*io\futures.py", line 274, in result*
*    raise self._exception*

*  File "C:\Users\Josep
Pablo\AppData\Local\Programs\Python\Python35-32\lib\async*
*io\tasks.py", line 239, in _step*
*    result = coro.send(None)*

*  File "C:\Users\Josep
Pablo\AppData\Local\Programs\Python\Python35-32\lib\site-*
*packages\pymata_aio\pymata_core.py", line 369, in start_aio*
*    self.log_output)*

*  File "C:\Users\Josep
Pablo\AppData\Local\Programs\Python\Python35-32\lib\site-*
*packages\pymata_aio\pymata_serial.py", line 50, in __init__*
*    writeTimeout=1)*

*  File "C:\Users\Josep
Pablo\AppData\Local\Programs\Python\Python35-32\lib\site-*
*packages\serial\serialwin32.py", line 31, in __init__*
*    SerialBase.__init__(self, *args, **kwargs)*
*TypeError: __init__() got an unexpected keyword argument 'writeTimeout'*

*C:\Users\Josep Pablo\AppData\Local\Programs\Python\Python35-32\Scripts>*

Josep

MrYsLab, 2016-01-09:
Ok, this error is because Pyserial was changed by its author and version 3.0 was loaded by my programs (I am awaiting a fix for pyserial). We need to force pyserial to be version 2.7. So first type:

`
pip uninstall pyserial
`

Then type:
`pip install pyserial==2.7
`

and then please try running blink again and let me know if that is successful.

JosepPablo, 2016-01-10:
I'm sorry but now i get it after try the uninstall:

*>pip uninstall pyserial*
*Fatal error in launcher: Unable to create process using '"'*

I tried also with pip3 with the same result.

Josep

MrYsLab, 2016-01-10:
The command should be:
`
 pip uninstall pyserial 
`

not:
pip uninstall pyserial*

If  the * was not part of your command line, then pyserial is not installed, so just try 
`
pip install pyserial==2.7 
`

JosepPablo, 2016-01-10:
I don't know why, but I needed execute:
python3 -m  pip uninstall pyserial
python3 -m  pip install pyserial==2.7
to uninstall pyserial 3.0 and install pyserial 2.7

After this, blink program was successfully executed.

Then I tried s2aio again.

It didn't open scratch and noticed me that didn't can open a path. I think
it's due to my user has an space, and this broke the path.

I tried with Snap, and opened it correctly, but the blocks didn't appear.

Then I put the command with the base path option with the next result:

With scratch option continues without opening and the same exception.
Snap opens, without the blocks. But a I could open it from the .xml files.

After this I think I can work with it.

Thank you very much for your effort!

Josep

MrYsLab, 2016-01-11:
Ok. I will close this issue, but please ask for help anytime.

closed 2016-01-11 by MrYsLab