MrYsLab/s2aio #6: Task was destroyed bit it is pending

ooledezma, 2016-03-20:
Hi I am looking forward to try this new extension. I followed the instructions on your Wiki but i get the following error when following the Starting S2AIO for scratch: 

C:\Users\ooled>s2aio

pymata_aio Version 2.11 Copyright (c) 2015 Alan Yorinks All rights reserved.

Using COM Port:com4

Initializing Arduino - Please wait... Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins

Task was destroyed but it is pending!
task: <Task pending coro=<PymataCore._command_dispatcher() running at c:\users\ooled\appdata\local\programs\python\python35\lib\sitepackages\pymata_aio\pymata_core.py:1232> wait_for=<Future pending cb=[Task._wakeup()]>>

I verified that i am running python 3.5.1.  

Can you please help me solve this issue?

MrYsLab, 2016-03-20:
Did you load firmataplus on your Arduino?

ooledezma, 2016-03-20:
Hi,

I tried to install firmataplus again and it worked. Sorry, I though i installed it the first time. It works now. Thank you.

ooledezma, 2016-03-20:
Two last questions, do i have to type s2aio in the CMD everytime I want to start scratch? Also, if i want to open an existing save project can I use import the extension on Scratch 2 offline?

MrYsLab, 2016-03-20:
You can start Scratch independent of s2aio by typing:
```
s2aio -c no_client

pymata_aio Version 2.11	Copyright (c) 2015 Alan Yorinks All rights reserved.

Using COM Port:/dev/ttyACM0

Initializing Arduino - Please wait... Auto-discovery complete. Found 30 Digital Pins and 12 Analog Pins

```

I typed the command you can see what to expect on your console.

Once you do this, you can open Scratch, and then do a File/Open of a previously created s2aio/Scratcha application (it has a .sb2 extension). The s2aio blocks will be loaded along with the project.

The only caveat is that you will need to close Scratch and s2aio separately.

If you want to load a different project into Scratch while s2aio is running, press the Scratch stop button, and then in the More Blocks area,  next to "s2aio" there is a drop down button. Select "Remove extension blocks". Now load the new program and you should be good to go.

I am closing this issue, but please feel free to continue the thread if you have any other questions.

closed 2016-03-20 by MrYsLab

ooledezma, 2016-04-05:
Hi

Do the arduino blocks get lost every time i close the program? I made a
game with arduino control but everytime i open it, the blocks go to
undefined.

MrYsLab, 2016-04-05:
It appears that the latest update to the  scratch editor has broken things. I reported this and the issue may be found here: https://github.com/LLK/scratch-flash/issues/1063

If the status changes I will you let you know or you can monitor that issue.

ooledezma, 2016-04-05:
> I reported this and the issue may be found here: LLK/scratch-flash#1063

Thank you for the help and support.

MrYsLab, 2016-04-05:
Apparently this bug was reported on the scratch site on March 22. Here is a link:
https://scratch.mit.edu/discuss/topic/188451/?page=1#post-1899612

It has been over 2 weeks since it was reported and still no fix!

ooledezma, 2016-04-05:
I just read through the discussion, I manage to find an older version of
scratch 444.2. I am using that in the mean time.

MrYsLab, 2016-04-07:
Version 445 of the editor seems to work for new projects that are saved and reloaded, but not for projects saved before version 444.

ooledezma, 2016-04-07:
I am going to give it a try and play with it.

*Oscar Ledezma* <http://www.linkedin.com/in/oscarledezmavazquez/>
*B.S. Physics*
California State Polytechnic University-Pomona, 2012
*M.S. Electrical Engineering*
California State University- Los Angeles
ooledezma@gmail.com
*(909) 753-3653*