MrYsLab/s2aio #7: Traceback error

diogojr2d, 2016-04-16:
Dear MrYsLab,

I'm using Linux (ubuntu), python3.5, and I was not able to use s2aio.
I already tried using -c no_client, -h, but I always get the same result.
When I try to start it, it shows me the following message:

> Traceback (most recent call last):
  File "/usr/local/bin/s2aio", line 9, in <module>
    load_entry_point('s2aio==1.8', 'console_scripts', 's2aio')()
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 351, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2363, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2088, in load
    entry = __import__(self.module_name, globals(),globals(), ['__name__'])
  File "/usr/local/lib/python3.4/dist-packages/s2aio/__main__.py", line 158
    async def kick_off(self, my_loop):
            ^
SyntaxError: invalid syntax

After reading the closed issues I also tried downgrading pyserial to 2.7, but it did not make any difference.
Could you help me?

MrYsLab, 2016-04-16:
@diogojr2d 
I think you might actually be using Python 3.4 and not Python 3.5, even though you installed Python 3.5. The reason I am saying this is because of the following from your traceback:
```
File "/usr/local/lib/python3.4/dist-packages/s2aio/main.py", line 158
async def kick_off(self, my_loop):
^
SyntaxError: invalid syntax
```
The packages were installed in the 3.4 directory and the async keyword was not valid in Python 3.4.
On Ubuntu 15.10 I needed to change the symbolic link in Ubuntu to point to the correct version of Python after installing Python 3.5.  Here is what my link was modified to:
```
:~$ ls -l /usr/bin/python3
lrwxrwxrwx 1 root root 9 Oct 27 14:18 /usr/bin/python3 -> python3.5
```
If you need to change the link, you will need to reinstall s2aio.

Please let me know if this solves your problem or if you need additional help.

MrYsLab, 2016-04-17:
I am closing this issue. If you are still have any issues, please place a comment on this thread and I will reopen the issue.

closed 2016-04-17 by MrYsLab