MrYsLab/s2aio #8: Sonar example not working

AbdouMec, 2016-07-25:
Hey MrYsLab..
I'd really like to thank you for your work, it really helped a lot in my project, I've been trying to use an ultrasonic sensor HC-SR04 but I just couldn't manage to get any readings from it. I tried using the example attached in the zip file I downloaded from your library, still not working. I'm not sure what am I doing wrong, as I managed to play with dc motors, IR sensors, etc. The conncetion is an accordance to the instructions in the refrence file, I'm kinda puzzled right now, any advice would be appreciated.
Regards.

MrYsLab, 2016-07-25:
Hi,
I just tested and I see no issues here. They way I connected the HC-SR04 to the Arduino was:

1. Connect VCC on the HC-SR04 to 5V on the arduino
1. Connect GND on the HC-SR04 to GND on the arduino
1. Connect Trig and Echo together on a bread board
1. Connect the common Trig and Echo to Pin 12 on the Arduino

To test your HC-SR04, could you please run the following Python script and see if you get any output:

```
from pymata_aio.pymata3 import PyMata3

# ping callback function
def cb_ping(data):
    print(str(data[1]) + ' centimeters')

# create a PyMata instance
board = PyMata3(2)

# configure 4 pins for 4 SONAR modules
board.sonar_config(12, 12, cb_ping)

while True:
    board.sleep(.1)

board.shutdown()
```
This script does not use Scratch at all and so simplifies things for test purposes.

Please let me know if this script gives you readings.

AbdouMec, 2016-07-27:
I tried running the script it seems like I'm missing something, as i get an import error that says " ImportError: No module named pymata_aio.pymata3". I'm using windows 10, and python 3.5.1. I know the HR-SR04 works as it gives reading when  used on Arduino IDLE, I might missed importing a certain library, this's what I get when  I try installing s2aio using pip:
Collecting s2aio
  Using cached s2aio-1.8.tar.gz
Collecting pymata-aio>=2.7 (from s2aio)
Collecting aiohttp>=0.19.0 (from s2aio)
  Downloading aiohttp-0.22.3.tar.gz (475kB)
    100% |################################| 481kB 202kB/s
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "c:\users\admin\appdata\local\temp\pip-build-6nu7np\aiohttp\setup.py", line 60, in <module>
        raise RuntimeError("aiohttp requires Python 3.4.1+")
    RuntimeError: aiohttp requires Python 3.4.1+

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in 

I'm  not sure if that where the problem lays, thanks for the fast reply.

MrYsLab, 2016-07-27:
It appears that pip may be linked to python2 and not python3 on your system and that is what is causing the error above. You may not have checked the box to add python 3.5 to your path when you installed python. [Here](https://github.com/MrYsLab/s2aio/blob/master/images/pythonInstall.png) is a screenshot. Before reinstalling Python 3.5 on your system, let's try something else.

Since the all other features in s2aio seem to be working for you, and you have verified the hc-sr04 is functional, perhaps it is your Scratch script. Could you please save the Scratch script you are using as a file and email me the .sb2 file so that I may test it here to see if I can reproduce your problem.  My email address is MisterYsLab@gmail.com

MrYsLab, 2016-08-01:
The user needs to properly install python 3.5. Once that is done and if the problem persists, this issue can be reopened.

AbdouMec, 2016-08-02:
Hi Alan!
So I uinstalled both versions of python, and reinstalled python 3.5.2. I used pip to install pyserial and pymata, the  issue now I can't even plug in my arduino to scratch, when I type in s2a_fm.py com3 to connect the board I get this error:
CMD window:C:\Users\user\Desktop\s2a_fm-master>s2a_fm.py com3  File "C:\Users\abdou\Desktop\s2a_fm-master\s2a_fm.py", line 57    print 's2a_fm version 1.5   Copyright(C) 2013-14 Alan Yorinks    All Rights Reserved '                                                                                         ^SyntaxError: invalid syntax

ps: I did upload FirmataPlus to board.

Date: Mon, 1 Aug 2016 04:30:14 -0700

MrYsLab, 2016-08-02:
You may already know this, but I will state it anyway. Python 2 and Python 3 are not compatible with each other. You may install both versions on your computer, but they are installed into different directories. Also, when you install packages, such as pyserial, if you need to use if for both python 2 and python 3, you need to install it for each version. There are separate copies kept for each version.

Ok, that being said, here are the steps I used to reinstall things. Please try following these steps I used and tell me if anything fails for you, which step and the error message.

**python 2.7 and s2a_fm**

1. I removed python 2 and 3 from my computer.
2. I downloaded and installed python 2.7.12 using all the defaults installation values.
3. It installed pytyon 2.7.12 in _C:\Python27_
4. I opened a command window and typed _cd c:\Python27_
5. In the C:\Python27 directory I typed _python_ and python started. I then closed python.
6. I then typed _cd Scripts_. This is where pip for python 2.7 is located.
7. I then typed: _pip install pyserial==2.7_
8. Pyserial 2.7 installed without issue.
9. I then typed _pip install PyMata_
10. PyMata installed without issue.
11. I then downloaded s2a_fm from github and unzipped the file. I went into the directory that contains s2a_fm.py and then typed: _c:\python27\python s2a_fm.py com5_
12. Your comport may be different than mine
13. s2a_fm started without issue.
14. I started scratch and brought up an s2a_fm project and was able to control an Arduino. 

**python 3.5 and s2aio**

1. Without removing anything, I downloaded python 3.5.2 and installed using the default install options, but I did check "_Add Python 3.5 to PATH_"
2. By checking this box, when typing _python_ in a command window it will automatically start python 3.5.2 and  if I type _pip_ in that window I will get the python 3.5 version of pip. I do not need to specify the paths when using python3, but I do when using python 2
3. If you want to see where the python 3 executables were installed type: _where python_ or _where  pip_
4. I then install s2aio by typing: _pip install s2aio_
5. It installed without issue and automatically installed pyserial and all the other packages it needs.
6. I then typed _s2aio_ and it started without issue

sobreira, 2016-12-07:
Hi MrYsLab.
I'd really like to thank you for great work, in fact we try to widespread the knownledge about your creation for lots of public school in Brazil.
We still use the Python 2.7 and I try to connect the HC-SR04.
I followed all steps above mentioned however Scratch or Snap can´t read any information from the sensor.

To make a test, I have installed the new Python and I did what you asked on your comment from25 Jul.
By this worked fine well, however, back to PyMata, Python 2.7 and s2a_fm, the Sonar not work.

For all the other other things, the system performs as expected.

I really appreciate your help on this.

Thanks!

Edson
hackeduca.com

MrYsLab, 2016-12-07:
Hi Edson,
   Are the problems you are having with the HC-SR04 on s2a_fm, s2aio or both?

   Are you able to get the HC-SR04 to work without using PyMata or pymata-aio?

Thanks,
Alan

sobreira, 2016-12-08:
Hi Alan, thanks for your prompt reply.
I am refering to s2a_fm - I just mentioned your above topic to make sure that the communication between FirmataPlus is OK.

I have also tested without Pymata and works fine.

MrYsLab, 2016-12-08:
Ok, let's try taking Scratch out of the equation for the moment. There are 2 examples and a wiring diagram [here](https://github.com/MrYsLab/PyMata/tree/master/examples/ping)

The examples are using Pin 12 for the sonar signal which you might need to match your setup and you might need to change the com port as well. Can you try each script and tell me if either one or both works?
Thanks.

sobreira, 2016-12-08:
t

sobreira, 2016-12-08:
Hi, Alan, I am not an expert on Python, so please to correct me if this is not what you have requested.

Following the test I did, first with:

- callback_ping_config_and_read.py com3  --- **Error**
- ping_config_and_read.py com3 --- **Error**

And just to make sure that Arduino is connected to the COM3, I typed also S2a_fm.py com3 and worked quite well.

[image]

sobreira, 2016-12-08:
Hi Alan,
I am still having problem in using the Sonar, do you have any other idea, I see that you closed the issue?
Thanks!

MrYsLab, 2016-12-08:
Sorry. I misunderstood. This is s2a_fm correct? If so, is your Scratch script similar to the one that works in s2aio?

sobreira, 2016-12-09:
No problem, thanks to reopened.
Yes is the s2a_fm (did I post in the wrong area?)
The Scratch script is exatcly the same

MrYsLab, 2016-12-09:
I am going to close this issue here and reopen it on s2a_fm. Here is a [link](https://github.com/MrYsLab/s2a_fm/issues/15) for that new issue.

closed 2016-12-09 by MrYsLab

MrYsLab, 2016-12-09:
@sobreira I am not sure if you saw this yet, but please check https://github.com/MrYsLab/s2a_fm/issues/15

and reply there.

Thanks

MrYsLab, 2017-03-05:
@sobreira I found an issue with reporting sonar values and it has been repaired with the latest release.

JosephMart, 2018-03-19:
Is this example fully functional now? I cannot seem to get it to work :/ I also have StandardFirmataPlus loaded on the board.

Connect VCC on the HC-SR04 to 5V on the arduino
Connect GND on the HC-SR04 to GND on the arduino
Connect Trig and Echo together on a bread board
Connect the common Trig and Echo to Pin 12 on the Arduino

```
from pymata_aio.pymata3 import PyMata3

# ping callback function
def cb_ping(data):
    print(str(data[1]) + ' centimeters')

# create a PyMata instance
board = PyMata3(2)

# configure 4 pins for 4 SONAR modules
board.sonar_config(12, 12, cb_ping)

while True:
    board.sleep(.1)

board.shutdown()
```

MrYsLab, 2018-03-19:
You need to install [FirmataPlus](https://github.com/MrYsLab/pymata-aio/wiki/Uploading-FirmataPlus-to-Arduino) and not StandardFirmataPlus. I do not know why the Firmata folks chose to use a name so close to the one I picked several years back, it has just added a lot of confusion for users.

This should be fully functional.

JosephMart, 2018-03-21:
Yep that was the my problem, thanks!

VinealShiloh, 2019-02-19:
> Is this example fully functional now? I cannot seem to get it to work :/ I also have StandardFirmataPlus loaded on the board.
>
> Connect VCC on the HC-SR04 to 5V on the arduino
> Connect GND on the [...]

I am getting unknown pin mode after running the above code

MrYsLab, 2019-02-19:
@VinealShiloh StandardFirmataPlus should not be used as stated on the [wiki installations for FirmataPlus](https://github.com/MrYsLab/pymata-aio/wiki/Uploading-FirmataPlus-to-Arduino).

For some reason, the author of Arduino Firmata chose to create a sketch with a name very similar to mine, even though he was aware that my sketch was developed and published several years before his.