MrYsLab/s2aio #9: Unstable startup "NoneType" object is not iteratable

Dradnats, 2016-08-11:
Hi MrYsLab,

Thanks for the nice and light weight application, it has been very useful for using in the teaching environment.  

Sometimes I get upon first connect the following error message:

TypeError: 'NoneType' object is not iterable

It was worse when I was attempting to start both the script and client, but has improved after using the  "-c no client" at startup.  Is is some sort of race condition in startup ?

Thanks,

Martin

full log: 1. unsuccessful 2. successful (both with -c no client) 
#########################################
C:\Users\dest>s2aio -c no_client

pymata_aio Version 2.15 Copyright (c) 2015-2016 Alan Yorinks All rights reserved.

Using COM Port:com5

Initializing Arduino - Please wait... list index out of range
Shutting down ...
--- Logging error ---
--- Logging error ---
Exception ignored in: <bound method Task.__del__ of <Task finished coro=<PymataCore._command_dispatcher() done, defined at c:\users\mthomaz\appdata\l
cal\programs\python\python35-32\lib\site-packages\pymata_aio\pymata_core.py:1250> exception=SystemExit(0,)>>
Traceback (most recent call last):
  File "c:\users\mthomaz\appdata\local\programs\python\python35-32\lib\asyncio\tasks.py", line 93, in __del__
  File "c:\users\mthomaz\appdata\local\programs\python\python35-32\lib\asyncio\futures.py", line 215, in __del__
  File "c:\users\mthomaz\appdata\local\programs\python\python35-32\lib\asyncio\base_events.py", line 1177, in call_exception_handler
  File "c:\users\mthomaz\appdata\local\programs\python\python35-32\lib\logging\__init__.py", line 1308, in error
  File "c:\users\mthomaz\appdata\local\programs\python\python35-32\lib\logging\__init__.py", line 1415, in _log
  File "c:\users\mthomaz\appdata\local\programs\python\python35-32\lib\logging\__init__.py", line 1425, in handle
  File "c:\users\mthomaz\appdata\local\programs\python\python35-32\lib\logging\__init__.py", line 1495, in callHandlers
  File "c:\users\mthomaz\appdata\local\programs\python\python35-32\lib\logging\__init__.py", line 855, in handle
  File "c:\users\mthomaz\appdata\local\programs\python\python35-32\lib\logging\__init__.py", line 986, in emit
  File "c:\users\mthomaz\appdata\local\programs\python\python35-32\lib\logging\__init__.py", line 908, in handleError
  File "c:\users\mthomaz\appdata\local\programs\python\python35-32\lib\traceback.py", line 100, in print_exception
  File "c:\users\mthomaz\appdata\local\programs\python\python35-32\lib\traceback.py", line 474, in __init__
  File "c:\users\mthomaz\appdata\local\programs\python\python35-32\lib\traceback.py", line 358, in extract
  File "c:\users\mthomaz\appdata\local\programs\python\python35-32\lib\traceback.py", line 282, in line
  File "c:\users\mthomaz\appdata\local\programs\python\python35-32\lib\linecache.py", line 16, in getline
  File "c:\users\mthomaz\appdata\local\programs\python\python35-32\lib\linecache.py", line 47, in getlines
  File "c:\users\mthomaz\appdata\local\programs\python\python35-32\lib\linecache.py", line 136, in updatecache
  File "c:\users\mthomaz\appdata\local\programs\python\python35-32\lib\tokenize.py", line 458, in open
  File "c:\users\mthomaz\appdata\local\programs\python\python35-32\lib\encodings\__init__.py", line 98, in search_function
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 954, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 887, in _find_spec
TypeError: 'NoneType' object is not iterable

C:\Users\dest>s2aio -c no_client

pymata_aio Version 2.15 Copyright (c) 2015-2016 Alan Yorinks All rights reserved.

Using COM Port:com5

Initializing Arduino - Please wait...
Arduino Firmware ID: 2.5 FirmataPlus32u4.ino
Auto-discovery complete. Found 30 Digital Pins and 12 Analog Pins

MrYsLab, 2016-08-11:
@Dradnats If you do a hard reset of the arduino (remove and replcace the usb cable) and then start s2aio, do you see the same behavior?

Dradnats, 2016-08-12:
@MrYsLab, it works when I perform:

1. Kill s2aio
2. Disconnect Usb
3. Power Cycle Arduino
4. Connect Usb
5. Start s2aio -no_client

If I do not do the Reset on Arduino, then it exits with NoteType error.

Probably the Arduino might be in a bad state to communicate with s2aio, but would be nice to get an error message indicating this problem. :)

MrYsLab, 2016-08-12:
@Dradnats The 32u4 reset circuitry sometimes behaves strangely. There is no way to do a software board reset on these boards (the Uno is different and does not exhibit this behavior) and that is why the power-on reset may be necessary.

Testing on a Leonardo, I am able run both with and without the no_client option and and am able to shutdown and restart repeatedly without issue. When the program exits, there are sometimes a list of exceptions, but the board does come up each time. 

If you use just use s2aio without any command line options, and shutdown by closing the Scratch editor, s2aio should then close automatically in about 2 seconds. If you do that, do you still have problems?

If you do, what Arduino are you using and is this an "official" Arduino or a clone?

Lastly, when you start s2aio, you should see something like this on the console:

```
pymata_aio Version 2.15	Copyright (c) 2015-2016 Alan Yorinks All rights reserved.

Using COM Port:/dev/ttyACM0

Initializing Arduino - Please wait... 
Arduino Firmware ID: 2.5 FirmataPlus32u4.ino
Auto-discovery complete. Found 30 Digital Pins and 12 Analog Pins
```

The firmware ID should 2.5. If you see something different, you may want to get the latest FirmataPlus versions [here](https://github.com/MrYsLab/pymata-aio/tree/master/FirmataPlus).

Dradnats, 2016-08-12:
@MrYsLab , 

My setup is using the littlebits arduino heartbeat, which has the an official leonardo device per my understanding.  I loaded "FirmataPlus32u4" loaded via arduino studio.

I repeated the same procedure, except used the s2aio with scratch client.

1. All started up ok first time, 
2. Exited scratch, errors appeared and s2aio exited.
3. started s2aio, scratch started up.
4. s2aio then reported errors, I have attached them below.

After exiting scratch, I'm unable to get a clean exit of s2aio, and upon next startup things are unstable.

[s2aio_Errors_v1.txt](https://github.com/MrYsLab/s2aio/files/415596/s2aio_Errors_v1.txt)

MrYsLab, 2016-08-12:
@Dradnats I don't have a littlebits Arduino so I won't be able to reproduce what you are seeing. Electrically, the littlebits module looks very much like an Arduino Leonardo, but I am confused by some of the output you provided. There seems to be "logging errors" being reported. Neither s2aio nor pymata-aio have logging enabled, so I am at a loss to know the origin of these error messages.  I have never seen this behavior before.

The only thing I can offer at this time is the workaround of removing and replacing power between s2aio sessions.

MrYsLab, 2016-08-14:
@Dradnats Since this issue seems to be specific to the littlebits module, nor can I reproduce what you are seeing on any other Arduino (tested with Uno, Leonardo, Mega 2560) I will be closing this issue. If you can provided additional information to help isolate the issue on the littlebits, I will be happy to re-open and pursue it.

closed 2016-08-14 by MrYsLab