MrYsLab/s2aio #10: S2aio - Task was destroyed

sobreira, 2016-12-07:
Hi, thanks for great development we have shared.
I am making a test in order to migrate from Python 2.7 to 3.5.2 and still using your development, however following all proposed tests, I have received the following message:

*********
[image]

I appreciate any help you can give me in this case.

Thanks!

MrYsLab, 2016-12-07:
Hi,
Could you please try running s2aio with the following command line and let me know if it still crashes:

```
s2aio -c no_client
```

Thanks,
Alan

sobreira, 2016-12-08:
Hi Alan, thanks

Another error was found in the process.
Following a screenshot with both errors

[image]

Thanks

Edson Sobreira

MrYsLab, 2016-12-08:
Thanks for the input. Could you please try the blink script located here. It must be executed with Python 3.5.

If you get a similar error as above, please change line 15 from
```
board = PyMata3()
```

to

```
board = Pymata3(arduino_wait=5)
```

Please let me know the result.

Thanks.

sobreira, 2016-12-08:
Hi Alan, I had no issue with

`board = PyMata3()`

[image]

However for the second suggestion I got the following:

[image]

MrYsLab, 2016-12-08:
I apologize, I should have said
```
board = PyMata3(arduino_wait=5)
```

The 'M' needs to be upper case. I am assuming that this will work if the first example worked. This is good news since it tells me that at least the pymata_aio portion of s2aio is functioning. Now on to s2aio. Could you please try the following:

1. Unplug your arduino from the USB port and the plug it back in.
2. Wait about 5 seconds.
3. Start s2aio with:
''''
 s2aio -c no_client
'''

Please let me know if you get different results from before.

sobreira, 2016-12-08:
`board = PyMata3(arduino_wait=5)`
Worked fine as well.

I followed the 3 steps you have mentioned, however show the following error:
[image]

sobreira, 2016-12-08:
quick question: do you prefer screenshot or copy and past the message here?

MrYsLab, 2016-12-08:
Hi Edson,
   Either way of presenting the information is fine.

   I believe I found the problem. The aiohttp library was updated and I am able to reproduce what you are seeing when I use the latest version.

   Since you may have both python 2 and python 3 installed on your computer, before changing the version of  aiohttp, first let's make sure of the version that you have on your system.

   From a command window, if you type:

'''
pip freeze
'''

You should see something like this:

[image]

You may have a different list of files which is OK, but I want to know if you see aiohttp listed. If it is listed, then we can load an older file by typing the following:

```
pip install aiohttp==0.22.4
```

If it is not listed, then pip is probably the Python 2 version and we will have to locate the python 3 version on your computer. But before doing that, let me know if you saw aiohttp listed from pip freeze and if so, if you were able to load version 0.22.4. If yes, also let me know if s2aio now is working for you. You only need to type s2aio without any options.

Thanks,
Alan

MrYsLab, 2016-12-08:
Version 1.10 addresses this issue. The aiohttp library was constrained to version 0.22.4.

sobreira, 2016-12-08:
Hi Alan, works like a charm.
Perfect. Thanks a lot
Connection is fine and also the Sonar worked as well

[image]

Edson Sobreira
hackeduca.com

MrYsLab, 2016-12-08:
Great. I am closing the issue. Please let me know if have any other questions.

closed 2016-12-08 by MrYsLab

davecraig, 2017-01-10:
Hi MrY, 

I think I'm hitting the same issue - but with s2aio 1.10. I've got a fresh Python install (3.5.2) and do the following:

C:\Users\Dave>py -m pip install s2aio
Collecting s2aio
  Downloading s2aio-1.10.tar.gz (531kB)
    100% |################################| 532kB 1.2MB/s
Collecting pymata-aio>=2.7 (from s2aio)
  Downloading pymata-aio-2.16.tar.gz
Collecting aiohttp==0.22.4 (from s2aio)
  Downloading aiohttp-0.22.4-cp35-cp35m-win_amd64.whl (131kB)
    100% |################################| 133kB 3.2MB/s
Collecting pyserial==2.7 (from pymata-aio>=2.7->s2aio)
  Downloading pyserial-2.7.tar.gz (122kB)
    100% |################################| 122kB 2.9MB/s
Collecting websockets (from pymata-aio>=2.7->s2aio)
  Downloading websockets-3.2-py33.py34.py35-none-any.whl
Collecting multidict<2,>=1.1.0 (from aiohttp==0.22.4->s2aio)
  Downloading multidict-1.2.2-cp35-cp35m-win_amd64.whl (134kB)
    100% |################################| 143kB 2.5MB/s
Collecting chardet (from aiohttp==0.22.4->s2aio)
  Downloading chardet-2.3.0.tar.gz (164kB)
    100% |################################| 174kB 2.3MB/s
Installing collected packages: pyserial, websockets, pymata-aio, multidict, chardet, aiohttp, s2aio
  Running setup.py install for pyserial ... done
  Running setup.py install for pymata-aio ... done
  Running setup.py install for chardet ... done
  Running setup.py install for s2aio ... done
Successfully installed aiohttp-0.22.4 chardet-2.3.0 multidict-1.2.2 pymata-aio-2.16 pyserial-2.7 s2aio-1.10 websockets-3.2

If I try uninstalling aiohttp and reinstalling it at the latest version (1.20) then s2aio fails to run because it depends on 0.22.4. Let me know if there's anything I can do to help,

Thanks!

Dave Craig.

MrYsLab, 2017-01-10:
Hi Dave,
Why are you updating aiohhtp and not using 0.22.4?

davecraig, 2017-01-10:
My apologies,

"Version 1.10 addresses this issue. The aiohttp library was constrained to
version 0.22.4."

I read this the wrong way round from what was intended. I realized shortly
after and deleted the comment (not quickly enough!). I had a different self
inflicted issue where I simply had StandardFirmataPlus and not FirmataPlus
installed on my Arduino.

It's now all working well, and I'll be retiring the well used s2_fm :-)

Apologies for the wasted time.

Thanks,

Dave

MrYsLab, 2017-01-10:
No problem. I am glad it is all working.