MrYsLab/s2aio #11: Sonar example not working - New

seteahconnoc, 2017-03-05:
Hello.

I am aiming to use a Raspberry Pi with an Arduino and sonar devices. At first, I wanted to use "snap4arduino". For testing purposes, I used my notebook successfully. Sadly, I found out, that it won 't work with a Raspberry Pi.

Gladly, I found your project "s2aio". So, I did the preparation. Some functions work; for example, I can manage a blinking of pin 13. But it is not possible to get distance values from the sonar device. In "Snap!", I just get an empty speech bubble.

Thank you for your work. I would be happy, if anyone could give me some advices. I would appreciate it, to keep the hints as simple as possible, for I am not quite experienced in using Python.

----------

Some specifications (others follow, if requested):

For testing purposes, I use a Debian notebook with:
3.16.0-4-amd64

s2aio -c snap with s2aio_Snap_base.xml

Arduino Firmware ID: 2.5 FirmataPlus32u4.ino according to:
https://github.com/MrYsLab/pymata-aio/wiki/Uploading-FirmataPlus-to-Arduino

/usr/local/lib/python3.5

Arduino Mega 2560

sonar device HC-SR04 connected like this:
https://github.com/MrYsLab/s2aio/wiki/Using-a-Sonar-Device

I checked out these issues:
https://github.com/MrYsLab/s2aio/issues/8
https://github.com/MrYsLab/s2a_fm/issues/15

MrYsLab, 2017-03-05:
I have verified what you are seeing. I will be investigating what the cause may be and will report back as soon as I have some information.

MrYsLab, 2017-03-05:
It appears that reporting any type of value in snap is not being reflected on the screen. I have verified that the data is being returned by the Arduino, so I need to see why Snap is behaving this way. Since s2aio uses common code for Scratch and Snap, and the reporters in Scratch function, the issue is either that something is broken in the Snap .xml code I provided, or something has changed in Snap to cause the condition.

Again, I will report back when I have some information.

MrYsLab, 2017-03-05:
I found the bug in the s2aio code for sonar reporting and repaired it. This fix is reflected in the latest release and has been placed in pypi as well.

Please let me know if this solves your issue.

seteahconnoc, 2017-03-06:
My issue is solved. Thank you very much for your immediate help.

closed 2017-03-06 by seteahconnoc