MrYsLab/s2aio #12: disconnet after save

Razzo1987, 2017-03-17:
Hi,
on Linux - Ubuntu 16.04 LTS after save scratch disconnet from arduino and s2aio on ather block became red.

MrYsLab, 2017-03-17:
Hi,
    I just tested on Windows 10 64 bit and Ubuntu 14.04 32 bit, and i can confirm what you are seeing on Ubuntu and that the problem does not exhibit itself on Windows. Since the s2aio code is exactly the same for both platforms, the problem is either in the Scratch editor or in Adobe Air. 

   That being said, I can offer a workaround, but it is less convenient to use.
Start s2aio with the -c no_client option:

```
s2aio -c no_client
```
Then manually start the scratch editor. You will then need to load a s2aio compatible project. These can be found here:
https://github.com/MrYsLab/s2aio/tree/master/s2aio/ScratchFiles/ScratchProjects

If you select one of the base projects using the File/Open option in the Scratch Editor, and then do a Save As, s2aio will not disconnect.

MrYsLab, 2017-03-17:
The version of Adobe Air being used with Linux is very old and Adobe Air no longer supports linux. Scratch does a miserable job of supporting Linux, and in fact, you cannot run Scratch on a 64 bit version of Linux 16.10. What I do is run s2aio in a VirtualBox Windows guest running on a linux host. 

Scratch does not notify extensions that the user has saved a project, so unfortunately, there is no way to reestablish connectivity that breaks as a result of the save, since the application is totally unaware that this has happened.

Razzo1987, 2017-03-19:
Hi,
this is the error in the terminal:

`razzo@razzo-N551JX:~$ s2aio

pymata_aio Version 2.16	Copyright (c) 2015-2016 Alan Yorinks All rights reserved.

Using COM Port:/dev/ttyUSB0

Initializing Arduino - Please wait... 
Arduino Firmware ID: 2.5 FirmataPlus.ino
Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins

Shutting down ...
Task was destroyed but it is pending!
task: <Task pending coro=<ServerHttpProtocol.start() running at /usr/local/lib/python3.5/dist-packages/aiohttp/server.py:246> wait_for=<Future cancelled>>
Exception ignored in: <generator object ServerHttpProtocol.start at 0x7f0423505db0>
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/aiohttp/server.py", line 331, in start
  File "/usr/lib/python3.5/asyncio/selector_events.py", line 566, in close
  File "/usr/lib/python3.5/asyncio/base_events.py", line 497, in call_soon
  File "/usr/lib/python3.5/asyncio/base_events.py", line 506, in _call_soon
  File "/usr/lib/python3.5/asyncio/base_events.py", line 334, in _check_closed
RuntimeError: Event loop is closed
Task exception was never retrieved
future: <Task finished coro=<PymataCore._command_dispatcher() done, defined at /usr/local/lib/python3.5/dist-packages/pymata_aio/pymata_core.py:1283> exception=SystemExit(0,)>
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/pymata_aio/pymata_core.py", line 1296, in _command_dispatcher
    next_command_byte = await self.read()
  File "/usr/local/lib/python3.5/dist-packages/pymata_aio/pymata_serial.py", line 155, in read
    await asyncio.sleep(self.sleep_tune)
  File "/usr/lib/python3.5/asyncio/tasks.py", line 516, in sleep
    return (yield from future)
  File "/usr/lib/python3.5/asyncio/futures.py", line 361, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib/python3.5/asyncio/tasks.py", line 296, in _wakeup
    future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 266, in result
    raise CancelledError
concurrent.futures._base.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/s2aio/__main__.py", line 859, in main
    the_loop.run_until_complete((s2aio.kick_off(the_loop)))
  File "/usr/lib/python3.5/asyncio/base_events.py", line 375, in run_until_complete
    self.run_forever()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 345, in run_forever
    self._run_once()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 1312, in _run_once
    handle._run()
  File "/usr/lib/python3.5/asyncio/events.py", line 125, in _run
    self._callback(*self._args)
  File "/usr/lib/python3.5/asyncio/tasks.py", line 299, in _wakeup
    self._step(exc)
  File "/usr/lib/python3.5/asyncio/tasks.py", line 241, in _step
    result = coro.throw(exc)
  File "/usr/local/lib/python3.5/dist-packages/pymata_aio/pymata_core.py", line 1341, in _command_dispatcher
    await self.shutdown()
  File "/usr/local/lib/python3.5/dist-packages/pymata_aio/pymata_core.py", line 1112, in shutdown
    sys.exit(0)
SystemExit: 0
razzo@razzo-N551JX:~$ 
`
with: 

> Start s2aio with the -c no_client option, then manually start the scratch editor.

it's work ;D

MrYsLab, 2017-03-19:
I am glad that the work around works.

closed 2017-03-19 by MrYsLab