MrYsLab/s2aio #13: Arduino Yun

Razzo1987, 2017-03-28:
Hi!
I have an Arduino Yun. With s4aio I can not connect to it.
there is a way?
I see this project: https://github.com/MrYsLab/s2aiomulti which, as I understand it, also manages more devices through IP and password (the routing map).
It is possible to use Yun with s2aio or i must use s2aiomulti ?

thanks a lot <3

MrYsLab, 2017-03-28:
Are you using the Yun as a Leonardo? If yes, you will need to load either the StandardFirmata sketch or FirmataPlus32u4 sketch onto the Yun. The YUN should look like a Leonardo to s2aio.

If you are trying to run s2aio on the Linux side of the Yun, you will need to install Python 3.5 or Python 3.6 on the Yun. I have never been successful in doing so and so I don't know how successful you might be if you are able to install it.

MrYsLab, 2017-03-29:
I am going to close this issue. If you have any other questions, you may add a comment to this issue or generate a new issue.

closed 2017-03-29 by MrYsLab