MrYsLab/s2aio #14: s2aio crashes

mayarblack, 2017-05-02:
Hi,
Please I need you help to resolve my problem:

First, when I upload the sktech "firmataPlus" to the Arduino UNO, I get this message: https://gist.github.com/mayarblack/568d534ab072ac96d7791d123f227d80.

second, when I run "s2aio -c scratch -l 5", scratch is opened, but s2aio crashes: https://gist.github.com/mayarblack/f6d6749d765e9075a80b03ad331a0ce1

I'm using Ubuntu 16.04 LTS 64 bits.
Python version: Python 3.5.2

thanks

MrYsLab, 2017-05-02:
@mayarblack What you are seeing when uploading firmartaPlus are compiler warnings and they can be safely ignored. In the past, Arduino turned warnings off by default, but the latest Arduiono software has them turned on.

I am surprised that you were able to get scratch to work on 64 bit Ubuntu. I have been unable to load the Adobe Air software and have only been able to do that with 32 bit. So I assume that you if you start Scratch by itself without s2aio, it comes up and runs normally.

Now to the crash. I would like to you try the following:

1. Unplug your Arduino USB cable and then plug it back in. Wait about 20 seconds for the lights on the Arduino to stop flashing.

2. Now start s2aio with: s2aio -l 5
You don't need the -c scratch option, but it should not harm anything either.

3. If you are still getting the crash, then repower the Arduino again, and start s2aio with:
s2aio -c no_client

4. Open up Scratch manually.

5. From the Scratch File menu, choose Open and load this file:

https://github.com/MrYsLab/s2aio/blob/master/s2aio/ScratchFiles/ScratchProjects/s2aio_base_FR.sb2

6. If you then go to the More Blocks tab in Scratch, you should see the s2aio blocks.

Please me know if either the first or second suggestion solved your problem.

Thanks.

MrYsLab, 2017-05-08:
Since I have not heard back from you, I am closing this issue. If you are still having an issue, please add a comment to this thread and I will be happy to reopen it and help you.

closed 2017-05-08 by MrYsLab