MrYsLab/s2aio #16: s2aio shuts down too quick on a slow computer

floquation, 2018-02-08:
Firstly, thank you for s2aio. It works wonderfully! But I did encounter an issue when using it on slow computers:

I got hold off some old laptops to use for educational purposes (CoderDojo). I found that s2aio correctly finds my Arduino, then it launches Scratch, and then it "immediately" shuts down again, before Scratch becomes usable.

I traced the problem to your Python "__main__.py" script in the `poll_watchdog` function. You wait 2 seconds there before shutting down s2aio when Scratch no longer sends keep-alive messages. Then what happens for me is that `current_time` is a large number, but `self.poll_time_stamp` is still 0, because Scratch was still starting for me. Consequently, s2aio shuts down, thinking that Scratch was shut down.

I found that it took my Scratch roughly 10s-15s before sending its first keep-alive message.

Would it be an idea to simply make the initial waiting time 30-60 seconds? I don't think there are any disadvantages to that (other than if the user closes Scratch within 30-60 seconds).
If you don't want to do that, a decent alternative might be to post a solution to this problem on the wiki: "Increase the number in line 787 of __main__.py considerably:"
`        await asyncio.sleep(2)`
(And, in case of posting it to a wiki, perhaps tell people where to find that file.)

MrYsLab, 2018-02-08:
I would be willing to add a command line option to allow you to set the value on line 787 to anything you
wish. The usage would be:
```
s2aio -s 60
```
This would change the default to 60 seconds. This would not affect existing users and should give you what you need.

floquation, 2018-02-08:
That'd do equally well. Thank you. (Saves me some hacking on every machine ;-) ).

MrYsLab, 2018-02-08:
The change is available in version v 1.12 and has been published to PyPi.

closed 2018-02-08 by MrYsLab