MrYsLab/s2aio #18: s2aio don't open Scratch

yanBrandao, 2018-02-16:
Hi folks,

I was navegating on google to learn something for teaching kids to use Arduino with sound tones. And I found your work. Congrats, its very nice.

So let's for the problem. I tried to install s2aio on Windows 8.1, with Python 3.6 and I had install everything including s2aio and forget the Scratch 2.0 :(

So when I looked to the wiki, you say to install everthing before s2aio, so yes. I did wrong.
And now when I type on prompt s2aio it doesn't open Scratch, You have any idea for what I need to do?

Best regards

MrYsLab, 2018-02-16:
When you start s2aio, please try adding the -s option to give Scratch time to start:
```
s2aio -s 10
```

This will give Scratch 10 seconds to start.  Please let me know if this works for you.

MrYsLab, 2018-02-23:
I am closing this issue since I have not received a reply. If you have further questions you can add them here and I will see them. If need be, we can reopen this issue.

closed 2018-02-23 by MrYsLab