MrYsLab/s2aio #19: unexpecting shutting down

kaszpirowskiMar, 2018-03-16:
mam problem z uruchomieniem s2aio. Podczas uruchomienia  pojawiają się komunikaty:

root@raspberrypi:/home/pi#  s2aio -c scratch -l 1 -p /dev/ttyACM0 -s 100

pymata_aio Version 2.19	Copyright (c) 2015-2017 Alan Yorinks All rights reserved.

Initializing Arduino - Please wait... 
Arduino Firmware ID: 2.5 FirmataPlus.ino
Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins

and after a few seconds, there appears a shouting down ...
At this time scratch 2 is running, (it's unhelpful - 100) the platform is rpi3,  python 3.6 .... what could be the reason?

kaszpirowskiMar, 2018-03-16:
I have a problem with running s2aio. messages appear during the listening:

root@raspberrypi:/home/pi# s2aio -c scratch -l 1 -p /dev/ttyACM0 -s 100

pymata_aio Version 2.19	Copyright (c) 2015-2017 Alan Yorinks All rights reserved.

Initializing Arduino - Please wait...
Arduino Firmware ID: 2.5 FirmataPlus.ino
Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins

and after a few seconds, there appears a shouting down ...
At this time scratch 2 is running, (it's unhelpful - 100) the platform is rpi3, python 3.6 .... what could be the reason?

MrYsLab, 2018-03-16:
Hi,
Are you using the version of Scratch 2.0  that is supplied with the latest versions of Raspbian, or are you running Scratch 2.0 downloaded from the Scratch site and are trying to connect to an Arduino?

MrYsLab, 2018-03-16:
Also, I just noticed that you are running as root. You should be able to run as a normal user.

MrYsLab, 2018-03-17:
Apparently Scratch 2.0 off-line from the Scratch site cannot be installed on a Raspberry Pi. It requires AdobeAir which is not compatible with the Raspberry Pi. If you wish to use the RPi GPIO with Scratch instead of an Arduino, please take a look at [s2-pi](https://mryslab.github.io/s2-pi/)

I am closing this issue since s2aio is incompatible with the Rpi. If you have any other questions, you may enter them here and I will see them.

closed 2018-03-17 by MrYsLab

kaszpirowskiMar, 2018-03-17:
after running from other user ... it started correctly ... scratch 2 installed from the distribution .... it would be good if you wrote in the wiki that root can not be run on s2aio ...

kaszpirowskiMar, 2018-03-17:
works ... the problem is that I'm using the -s parameter if I set it to 1000 so at this time I can control the arduino (starting the blinking LED) ... if the time is over it turns off s2aio ....

MrYsLab, 2018-03-17:
How did you install AdobeAir on the RPi? I would be curious as to the procedure.
Since I cannot run s2aio on my Raspberry Pi, I do not know why things are timing out for you. To get around this, use the [no_client feature] ( https://github.com/MrYsLab/s2aio/wiki/Starting-And-Customizing-s2aio-for-Scratch#client-option) and start Scratch manually and then load a project file in the language of your choice from this directory: https://github.com/MrYsLab/s2aio/tree/master/s2aio/ScratchFiles/ScratchProjects

kaszpirowskiMar, 2018-03-17:
I did not install any additional ...
- python 3.6 from sources
- scratch 2.0 (offline)
- s2aio from pip indicated to python 3.6 - default is to the repository 3.4 and I could not get rid of it in the easy way so I have 2 versions.

As for AdobeAir, I have heard that it does not support PiZero and it is not possible to run scratch 2.0 at all

MrYsLab, 2018-03-17:
I still don't understand how you were able to install Scratch 2.0 off-line since it requires AdobeAir. The version of Scratch that is installed with Raspbian is not compatible with s2aio, and therefore s2aio cannot open Scratch with the Raspbian version. Running s2aio will just try to find Scratch and eventually time out and exit.

MrYsLab, 2018-03-21:
Version 1.14 will allow s2aio to work with Raspberry Pi. Usage:
```
s2aio -r true
```

kaszpirowskiMar, 2018-03-21:
ok ... but unfortunately on my RPI I cooked gpio .... now it serves as a driver for Arduino with which to connect from the notebook

MrYsLab, 2018-03-21:
If your USB connectors are still working on the RPi, plug your arduino into one and s2aio should be able to control the arduino.

kaszpirowskiMar, 2018-03-23:
and this is just happening ... only wireless communication is needed for luck ...