MrYsLab/s2aio #20: irda in s2aio

kaszpirowskiMar, 2018-03-17:
has anyone tried to set up irda (lirc) communication in s2aio? it's about pilot control ....

MrYsLab, 2018-03-17:
I know very little about IRDA. s2aio uses modified StandardFirmata to control an Arduino and therefore needs that specific sketch to be loaded on the Arduino. My guess is that s2aio would have to be drastically modified (and probably the sketch as well) to work with IRDA.

closed 2018-03-17 by MrYsLab

kaszpirowskiMar, 2018-03-17:
and if there is any other way of communication (wifi serial, bluetuth or other) ... that can handle firmataplus?

MrYsLab, 2018-03-17:
You could use Bluetooth, or any other device that provides a com port id and can be configured to a serial baud rate of 57600. This [artticle](http://www.instructables.com/id/Wireless-Arduino-Control-Using-the-BeagleBone-Blac/) explains how to configure a Bluetooth dongle to work with an Arduino.

kaszpirowskiMar, 2018-03-17:
in the package with firmata plus there was also firm plus wifi ..... does it also have the same properties as the company plus plus additionally wifi communication (eg service module wirles esp8664) ...?

kaszpirowskiMar, 2018-03-17:
sorry insorry mistake ... no firmdata plus wifi only standard wifi ... question whether the functionality of wifi (espion64 module) can be added to the plus version ....?

MrYsLab, 2018-03-17:
There may be some confusion as to what is supplied with the standard Arduino software and what sketches I supply in [FirmataPlus](https://github.com/MrYsLab/pymata-aio/raw/master/FirmataPlus/libraries.zip).

Here are the packages that are supplied with the standard Arduino software:
[image]

And here is what is in my FirmataPlus package:
[image]

My sketches do not support wifi nor the esp modules. I have no plans for adding functionality to the existing sketches at this time.

kaszpirowskiMar, 2018-03-17:
... is there any chance for firmataplus to swap the usb cable in communication on something radio? .. communication needed to control the robot from arduino on board ..... which would be controlled with scratch2.0

MrYsLab, 2018-03-17:
I am not sure how you are defining "radio", but as explained above, you can use Bluetooth. At this time no changes are being planned because it is not clear if the Scratch folks are going to support Scratch 2.0 and its extensions once Scratch 3.0 is released later this year.

kaszpirowskiMar, 2018-03-17:
in the plus format which is used in communication with strache20 ...? if so, I am tempted to buy the right BT module .....

MrYsLab, 2018-03-17:
All the devices I used are listed in [this article](http://www.instructables.com/id/Wireless-Arduino-Control-Using-the-BeagleBone-Blac/). Please note that s2aio has NEVER been tested with Bluetooth, so proceed at your own risk. I may not be able to supply support if things do not work as you expect.

kaszpirowskiMar, 2018-03-17:
ok ... I'm screwing up