MrYsLab/s2aio #21: le system ne peut trouver le fichier c:\program

nonoecolo, 2018-05-12:
Hello !!
I have followed the installation wiki but i have this message .. Any idea ...???
[image]

MrYsLab, 2018-05-12:
There are a few things you can try:

1. Start s2aio with the -s  option. This will allow Scratch additional time to initialize. I would start with a large number, such as 15, and if that works you can try smaller and smaller numbers to see where it fails.

```
s2aio -s 15
```

If that doesn't work, enter:
```
s2aio -v
```
This will print the directory where the s2aio files should be installed. When I run this command, here is what I get:
[image]

If I then go to ScratchFiles/ScratchProjects, I see the following files:
[image]

If you see a similar set of files in your directory, then you can try starting things manually by using the -c option:
```
s2aio -c no_client
```
After you see the "autodiscovery complete" message appear, start Scratch and using File/Open, select one of the files to load the blocks. For example for French blocks, you would select s2aio_base_FR.sb2.

The blocks should now be loaded.

Please let me know if any of this works for you.

MrYsLab, 2018-05-14:
Have you had any success? If not, please let me know if you tried both suggestions above.

MrYsLab, 2018-05-16:
Since I have not received a response, I am closing this issue. If you are still having difficulties, you can add a comment to this thread, and I will be happy to reopen the issue.

closed 2018-05-16 by MrYsLab