MrYsLab/s2aio #22: SyntaxError: invalid syntax | Python 3.7 | macOS 10.13.5

bestia-io, 2018-07-02:
The following error occurs no matter the "s2aio" command I input to Terminal.  I saw a similar issue previously closed... Any ideas?

> Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.7/bin/s2aio", line 7, in <module>
    from s2aio.__main__ import main
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/s2aio/__main__.py", line 28, in <module>
    from aiohttp import web
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/aiohttp/__init__.py", line 10, in <module>
    from .protocol import *  # noqa
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/aiohttp/protocol.py", line 17, in <module>
    from .helpers import reify
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/aiohttp/helpers.py", line 26
    ensure_future = asyncio.async
                                ^
SyntaxError: invalid syntax

MrYsLab, 2018-07-02:
The version of aiohttp that is being used by s2aio is not compatible with python 3.7. I am investigating to see if I can convert over to a newer version of aiohttp that is compatible.  As a result, at this time, s2aio is compatible with Python 3.5.x, 3.6.x, but not 3.7.

MrYsLab, 2018-07-02:
@bestia-io This should now be fixed with version 1.15. Thanks for letting me know about the issue.

closed 2018-07-02 by MrYsLab