MrYsLab/s2aio #24: S2aio shutsdown a few minutes after starting

amurray22, 2018-08-08:
I have recently installed S2aio and Scratch 2 offline editor following the directions in your write-up and it installs and runs for a few minutes but then the S2aio shuts down, even though I have scratch open and am using it with the extra blocks. I have tried the -s option with upto 120 but that did not resolve the issue. Scratch does start and I can use it with the Arduino perfectly, but the S2aio just shuts down after a few minutes.

Noy very familar with python coding so I can't give you much more for troubleshooting. I am using it on a Mac.

Thanks
Andy

MrYsLab, 2018-08-08:
Could you please try starting s2aio with the -c no_client option then start scratch manually and load the following project: https://github.com/MrYsLab/s2aio/blob/master/s2aio/ScratchFiles/ScratchProjects/s2aio_base.sb2

If that works can you tell me which version of pyserial you are using?

amurray22, 2018-08-09:
Ok, that seems to have worked. 

I have pyserial 3.4 installed. 

Should I just continue to use these new commands and base Scratch file?

Thanks
Andy

MrYsLab, 2018-08-09:
You can run in the no_client  manual mode, but the auto mode should work. What is happening is that Scratch is not requesting poll data often enough and a watchdog timer is firing and closing things down, thinking that Scratch has shut down. The no_client mode does not use the watchdog timer and so things continue to run. So either there is a problem with the latest version of Scratch or with pyserial. Unfortunately I don't have access to a Mac and can't test the behavior you are seeing.

When things shutdown do you have a Scratch project running? If yes, if you start s2aio and just let it sit, does it shutdown if not running any s2aio blocks?

amurray22, 2018-08-09:
Ok, thanks.

What I was being was that I would start S2aio and Scratch and then start using the blocks. It would suddenly ‘lose’ connection ie- the indicator light in Scratch blocks would change to red and anything I was doing with the Arduino would stop. If I simple went back to my terminal and started S2aio again things would be find again for about that same time. The watch dog makes perfect sense. I’m just not sure how they interact so not sure why it never gets the message Scratch is running.

Andy

MrYsLab, 2018-08-09:
If it is only shutting down when you are controlling the Arduino, then perhaps you have a tight forever loop in your Scratch script. Try adding a wait block with short time period, .1, .01, etc. to the loop and see if that helps.

MrYsLab, 2018-08-09:
@amurray22 Are you still having issues, and if so, please let me know what issues remain. If not, can I close this issue?

amurray22, 2018-08-10:
The work around you provided works well, but the standard method still causes the timeout.

MrYsLab, 2018-08-10:
One last question do your have firmataplus installed on the arduino? If yes, are you using this version:
https://github.com/MrYsLab/pymata-aio/blob/master/FirmataPlus/libraries.zip. StandardFirmata should also work. The version of firmata should be printed on the console as the "firmware id" when you start s2aio, could you please let me know which version is being shown:

[image]

amurray22, 2018-08-11:
Hello,

Here is the response to pip

abstract-rendering==0.5.1
alabaster==0.7.6
anaconda-client==1.2.1
appnope==0.1.0
appscript==1.0.1
argcomplete==1.0.0
astropy==1.0.6
Babel==2.1.1
beautifulsoup4==4.4.1
bitarray==0.8.1
blaze==0.8.3
bokeh==0.10.0
boto==2.38.0
Bottleneck==1.0.0
cffi==1.2.1
clyent==1.2.0
colorama==0.3.3
conda==3.18.8
conda-build==1.18.2
conda-env==2.4.5
configobj==5.0.6
cryptography==1.0.2
cycler==0.9.0
Cython==0.23.4
cytoolz==0.7.4
datashape==0.4.7
decorator==4.0.4
docutils==0.12
dynd==9b63882
fastcache==1.0.2
Flask==0.10.1
greenlet==0.4.9
h5py==2.5.0
idna==2.0
ipykernel==4.1.1
ipython==4.0.1
ipython-genutils==0.1.0
ipywidgets==4.1.0
itsdangerous==0.24
jdcal==1.0
jedi==0.9.0
Jinja2==2.8
jsonschema==2.4.0
jupyter==1.0.0
jupyter-client==4.1.1
jupyter-console==4.0.3
jupyter-core==4.0.6
llvmlite==0.8.0
lxml==3.4.4
MarkupSafe==0.23
matplotlib==1.5.0
mistune==0.7.1
multipledispatch==0.4.8
nbconvert==4.0.0
nbformat==4.0.1
networkx==1.10
nltk==3.1
nose==1.3.7
notebook==4.0.6
numba==0.22.1
numexpr==2.4.4
numpy==1.10.1
odo==0.3.4
openpyxl==2.2.6
pandas==0.17.1
path.py==0.0.0
patsy==0.4.0
pep8==1.6.2
pexpect==3.3
pickleshare==0.5
Pillow==3.0.0
ply==3.8
psutil==3.3.0
ptyprocess==0.5
py==1.4.30
pyasn1==0.1.9
pycosat==0.6.1
pycparser==2.14
pycrypto==2.6.1
pycurl==7.19.5.1
pyflakes==1.0.0
Pygments==2.0.2
pyOpenSSL==0.15.1
pyparsing==2.0.3
pytest==2.8.1
python-dateutil==2.4.2
pytz==2015.7
PyYAML==3.11
pyzmq==14.7.0
qtconsole==4.1.1
redis==2.10.3
requests==2.8.1
rope-py3k==0.9.4.post1
scikit-image==0.11.3
scikit-learn==0.17
scipy==0.16.0
simplegeneric==0.8.1
six==1.10.0
snowballstemmer==1.2.0
sockjs-tornado==1.0.1
Sphinx==1.3.1
sphinx-rtd-theme==0.1.7
spyder==2.3.8
SQLAlchemy==1.0.9
statsmodels==0.6.1
sympy==0.7.6.1
tables==3.2.2
terminado==0.5
toolz==0.7.4
tornado==4.3
traitlets==4.0.0
ujson==1.33
unicodecsv==0.14.1
Werkzeug==0.11.2
xlrd==0.9.4
XlsxWriter==0.7.7
xlwings==0.5.0
xlwt==1.0.0

It does not look like I have the same modules as you listed. Not sure if they are listed some other way.

The version of Scratch is 461 of the offline scratch 2.0

I am not using a genuine board, it is an OSEPP Uno R3 Plus.

I have tried the following combination of running S2aio and scratch

1 -Run Scratch with no special project (just opening Scratch) and no blocks or commands used, Arduino connected to USB, run S2aio (no switches/commands) s2aio starts successfully, finds comm port and arduino then shuts downs in about 1 minute

2- Run Scratch with standard Scratch-Arduino project file (S2aio-base), Arduino connected to USB, run S2aio (no switches/commands) s2aio starts successfully, finds comm port and arduino. Scratch shows green indictor in More Blocks section. Still I did not use and blocks or commands, just left it running. S2aio still shutdown but much later, maybe 3-4 minutes

3-Run Scratch with standard Scratch-Arduino project file (S2aio-base), Arduino connected to USB, run S2aio (no switches/commands) s2aio starts successfully, finds comm port and arduino. Scratch shows green indictor in More Blocks section. I start using the arduino based blocks and it works. Seems to be fine now as long as I am using commands it does not time out. If I stop for more then a few minutes then it does shutdown S2aio. I guess I just need to be using the connection more often.

I appreciate you spending the time to troubleshoot.

Andy

> On Aug 10, 2018, at 10:21 AM, Alan Yorinks <notifications@github.com> wrote:
> 
> pip freeze

amurray22, 2018-08-11:
Here is what my terminal says

pymata_aio Version 2.22	Copyright (c) 2015-2017 Alan Yorinks All rights reserved.

Using COM Port:/dev/tty.usbserial-A602QNHH

Initializing Arduino - Please wait... 
Arduino Firmware ID: 2.5 FirmataPlus.ino

Thanks
Andy

amurray22, 2018-08-11:
> @amurray22 <https://github.com/amurray22> Andy, Python separates out its environments for python 2 vs 3. So you might need to do
>
> pip3 freeze
> The s2aio files should then appear.
>
> If this list contains the files I am looking for can you please attach the list?
>
> A few questions:
>
> Are you running s2aio in a standard environment vs a python virtual environment?
> Do you start s2aio from a terminal command prompt or are you using some Mac facilities, or perhaps something via anaconda or jupyter?
> It looks like you are using anaconda. Have you installed it for both python2 and 3?
> Do you have any other programs running beside s2aio when you ran your 3 test scenarios?
> When you installed s2aio, did you do so from a terminal with the command:
> sudo pip3 install s2aio
> if not, how did you install it?
>
> The behavior you are seeing is something I have not encountered before and it definitely points to a loss of communication between Scratch and s2aio over the USB/serial port. When the green/red connectivity indicator in Scratch turns red, it means that Scratch has not received a reply to a poll it has sent (it sends about 30 polls per second). I am not able to reproduce what you are seeing in Windows or on linux, so I do not know yet if the problem is tied to the MAC os, MAC hardware, the USB cable, Scratch for the MAC, or the AdobeAIR library.
> Thanks.

I have been following this instructable - 

I already had python installed so I did not follow that step, but everything else and am using the normal Mac terminal window. So the command I used to install S2aio was 'sudo pip3 install s2aio’

This looks more like what you were looking for. Its from the pip3 command

aiohttp==3.3.2
async-timeout==3.0.0
attrs==18.1.0
chardet==3.0.4
idna==2.7
multidict==4.3.1
pymata-aio==2.23
pyserial==3.4
s2aio==1.16
websockets==6.0
yarl==1.2.6

I’m not sure what anaconda is. How can I check if its associated with python 3?

When I did my test I probably had firefox, Mac Mail, Scratch2

Let me know what else I can check/test.

Andy

MrYsLab, 2018-08-11:
@amurray22 Anaconda is used to install Jupyter. If  you start Python3 from a terminal window, it will announce it is an Anaconda installation similar to this:
```
$ python3
Python 3.6.5 |Anaconda, Inc.| (default, Apr 29 2018, 16:28:17) 
[GCC 7.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```
If you don't see Anaconda in the heading, then it is a standard python installation.

If you can close any open windows (such as Firefox, Mac Mail, and Scratch2) and retry your #1 test to see if that affects the results.

When you ran in no_client mode, did the Scratch connectivity indicator ever flash between red and green?

amurray22, 2018-08-11:
Ok, looks like I do not have that.

Python 3.7.0 (v3.7.0:1bf9cc5093, Jun 26 2018, 23:26:24) 
[Clang 6.0 (clang-600.0.57)] on darwin
Type "help", "copyright", "credits" or "license" for more information.

I re-did test #1 with all other apps closed and it still shutdown after a minute or two. I then did the same setup but ran S2aio with the -c none parameter and it stayed connected. Probably left it running about 5 minutes. IN both cases I did nothing in scratch, other than have it open.

Andy

MrYsLab, 2018-08-11:
Ok, thanks. Looking things over, I was mistaken and the problem is not on the serial/Arduino side of things, but either in the s2aio watchdog or scratch/HTTP server.  I just installed python 3.7 on a Linux machine that can run Scratch and was hoping it would crash, but no luck. Usually, Linux and the MAC behave pretty similarly, 

Scratch sends poll requests at about 30 times/second and each time a poll is received, a poll reply is sent to Scratch and the time the poll was received is saved. The poll watchdog checks to see if a poll was not received in greater than 1 second which would result in closing s2aio because no poll requests means that Scratch was closed or crashed. To check the timeframes, I modified a local copy of the code just to verify that maybe somehow I might have miscalculated things with the watchdog, but the longest time period between polls was .02 seconds. 

The only other variable I can think of is that I recently went to a newer version of the aiohttp webserver (Scratch uses HTTP to communicate with s2aio). I checked the issues list for aiohttp and did not find any problems related specifically to the MAC. 

So at this point, I am out of ideas as to why you are seeing the crash behavior.

If you wish to print the timestamps in the poll watchdog to see if it ever exceeds 1 second, here is a procedure to do so:

1. You will need to find where the __main__.py file was installed by pip. The file name has to leading and trailing underbar characters, but they do not appear here because of mardown formattintg. To do so in a terminal window, type: pip3 show s2aio. 
```
$ pip3 show s2aio
Name: s2aio
Version: 1.16
Summary: A Scratch 2.0 (Offline) Hardware Extension for Arduino
Home-page: https://github.com/MrYsLab/s2aio/wiki
Author: Alan Yorinks
Author-email: MisterYsLab@gmail.com
License: GNU General Public License v3 (GPLv3)
Location: /usr/local/lib/python3.7/site-packages
Requires: aiohttp, pymata-aio
Required-by: 
```
The line labeled **Location** will show the path to where pip installs things. 

2. If you cd to that path and then cd to s2aio, you should see a file named **__main__.py** (with the underbars)

```
afy@afy-VirtualBox:~$ cd /usr/local/lib/python3.7/site-packages/s2aio
afy@afy-VirtualBox:/usr/local/lib/python3.7/site-packages/s2aio$ ls
configuration  __init__.py  __main__.py  __pycache__  ScratchFiles  Snap!Files
afy@afy-VirtualBox:/usr/local/lib/python3.7/site-packages/s2aio$ 
```
3. Copy this file to another name as a backup copy in case things go wrong.
4. You will need to add a line to the poll_watchdog method.
The code for the poll_watchdog in __main__.py is:
```
async def poll_watchdog(self):
        """
        This method is enabled for scratch clients. It monitors to see if polls are no longer being
        sent, and if so, shuts down the server. It waits 2 seconds before shutting down.
        :return: None
        """
        await asyncio.sleep(self.sleeper)
        while True:
            await asyncio.sleep(1)
            current_time = self.loop.time()
            print(current_time - self.poll_time_stamp)
            if current_time - self.poll_time_stamp > 1:
                await self.board.send_reset()

                for t in asyncio.Task.all_tasks(self.loop):
                    t.cancel()
                self.loop.run_until_complete(asyncio.sleep(.1))
                self.loop.stop()
                self.loop.close()
                sys.exit(0)
```
5. Add the print line shown above to the file.
6. Save and start s2aio. After about 10 or 11 seconds you should see the times being printed to the console:
```
~$ s2aio

pymata_aio Version 2.22	Copyright (c) 2015-2017 Alan Yorinks All rights reserved.

Using COM Port:/dev/ttyACM0

Initializing Arduino - Please wait... 
Arduino Firmware ID: 2.5 FirmataPlus.ino
Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins

0.017752075000316836
0.014159413000015775
0.015115756999875885
0.02438324000013381
0.015175943000031111
0.01878114299961453
0.029103828000188514
0.02264679400013847
0.00045987300018168753
```
If you try this, let me know if the time printed exceeds 1 second right before the crash.

MrYsLab, 2018-08-19:
Hi Andy,
    Thanks for taking the time to look at this and getting back to me. If you like, we can try one more thing. We can extend what constitutes a timeout from >1 to >2 seconds in the poll_watchdog method

In poll_watchdog if you change the line that reads:
```
            if current_time - self.poll_time_stamp > 1
```
to:
```            
            if current_time - self.poll_time_stamp > 2
```
Please let me know if that solves the problem. If it does, then reading the time on your mac is slightly different than for other computers in how things are returned. I will then make the above change permanent. 

Thanks.

amurray22, 2018-08-20:
Ok, I tried changing it to 2 and it is fine for a while and then it quickly increases the time until it passes 2 seconds. I then changed it to 10 and noticed an odd pattern. Once it seems to go to long time intervals they become exactly 1 second and never past 10. after 9.xxxx it goes to a small decimal number and counts up again. here is an example:

^CAndrews-MBP:~ andrewmurray$ s2aio

pymata_aio Version 2.22	Copyright (c) 2015-2017 Alan Yorinks All rights reserved.

Using COM Port:/dev/tty.usbserial-A602QNHH

Initializing Arduino - Please wait... 
Arduino Firmware ID: 2.5 FirmataPlus.ino
Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins

0.008967703999999799
0.005716620000001171
0.008372736000001879
0.0062769910000000095
0.009215216000001192
0.009764545999999541
0.010003685000000928
0.004541686999999683
0.008655355000001919
0.008642838000000097
0.010508596999997621
0.009956954999999823
0.00691011099999983
0.010332334000000998
0.010378797000001327
0.8729630750000013
1.873148208
2.873325340000001
3.8734831160000027
4.873653824000002
5.873850828000002
6.874037536000003
7.874114924000004
8.874228285999997
9.874418714
0.8448138470000046
1.8449271260000017
2.8450142140000025
3.8451642090000036
4.8453066460000045
5.845505213999999
6.845627059999998
7.845744093
8.845884081999998
9.846033925
0.8129779930000041
1.8131577960000058

The other strange thing is that these numbers are written to the screen faster then the one second that it should be between the numbers. Not sure if they are queued up before printed or something.

So if I leave the watchdog at >10 it stays online, but I guess this is just like using the switches to disable the watchdog.

Andy

MrYsLab, 2018-08-21:
Andy,
    I tried adjusting the watchdog timeout above 1 on my system in hopes that I might see similar behavior to what you are seeing, and it did not change anything on my computer. The largest time period reported between poll receipts from scratch was .03 seconds. So this did not offer any clues as to what might be happening for you.

   Based on your print data, it looks like one of the following might be occurring:

1. Scratch stops sending out poll requests.
2. Somehow the request for the current loop time returns values that are not accurate, and therefore the watchdog expires.
3. The HTTP server, aiohttp is somehow not passing on the poll requests from Scratch.

None of these make sense since there are no similar symptoms on either Windows or Linux. Again, since I have users running s2aio on the Mac, I am totally stumped as to what is happening on your system.

I will keep thinking about possible causes, but I would suggest you run the program in no_client mode.
I am sorry I have no better solution at this time.

amurray22, 2018-08-21:
No problem. I have a easy workaround so its just a fun exercise in troubleshooting. The thing that does make sense to me (not that I understand much of this) is that the number keeps coming, albeit longer. I’d think if it stops or loses comms that all of a sudden their would be no number, or is there something else that would have the number created regardless?

Andy

MrYsLab, 2018-08-22:
Andy,
    The way things work with asyncio is that asyncio uses its own event loop. You can think of the event loop as an intelligent scheduler, kind of like a mini operating system (but very limited in what it can do). 

It uses a form of cooperative scheduling, that is, a function can return control to event loop by using the await keyword when making a function call. The event loop then continues running from where it last left off while waiting for the "awaited" function to return.

So in the case of s2aio, the poll_watchog method calls asyncio.sleep(1) as the first instruction in its while True loop. This returns control back to the event loop that will allow other portions of the program to run while the event loop is waiting for the 1 second time to expire.  During this "waiting period" at some point, control is returned to the aiohttp web server. When a poll request comes in from Scratch (it generates about 30 per second), it then will call the poll method to process the request. The poll method saves a time stamp of when the poll request was received and then generates an HTTP response back to Scratch. (BTW, If Scratch does not receive a reply, it eventually will turn the connectivity indicator red. Scratch is still running at this time, and can still process commands.). Since nothing else is going on in s2aio when you just start s2aio and let it sit idle, there can be nothing blocking the call to aiohttp.

When the 1-second timer finally expires, control is returned back to poll_watchdog, it compares the current time to the last time that a poll was received. If it is greater than 1 second, it assumes that the user closed Scratch, and the s2aio is shut down.

It is a mystery to me why you are seeing the printed times increase the way they do.

The reason things keep running until you hit the maximum comparison value is that Scratch does not crash if it does not receive an HTTP reply to its HTTP request, but just keeps going on. The comparison value is one that s2aio uses to determine how long a poll has not been received and then bails out when the number is exceeded. 

Again, since I cannot recreate the behavior you are experiencing, I cannot venture a guess as to what is actually happening.

Thanks for your patience through all this. If I can come up with an explanation, I will get back to you.