MrYsLab/s2aio #26: Path on Windows 10 does not accept spaces

IainColledge, 2019-03-16:
When installing on Windows 10 the client cannot be found and the error states the path is "c:\program".

Installing to C:\ results in a path error of "c:\scratch" as the scratch 2 install path is "c:\Scratch 2".

Renaming this to C:\scratch2\scratch2.exe and setting the path in the configuration file results in the client starting up although it shuts down very fast.

-c no_client allows s2aio to stay up but the blocks are not installed under scratch 2.

Please advise.

MrYsLab, 2019-03-16:
I just reinstalled both s2aio and the Scratch 2.0 editor on Windows 10 and I am not seeing the error you are reporting.

When installing Scratch from https://scratch.mit.edu/download/scratch2
I accept the defaults:

[image]

I am using pip to install s2aio:

```
pip install s2aio
````

When I start s2aio, I do not see any errors. 

I do not understand your first statement:

> When installing on Windows 10 the client cannot be found and the error states the path is "c:\program".

Could you please explain what steps you are taking to cause this?

IainColledge, 2019-03-16:
Thanks for the reply.

I removed Scratch 2 and installed as per default.

To keep things simple am not using a virtual env so installing under the system python, version 3.7.2.

I'm also using pip 19.0.3

I install using admin powershell.

When running s2aio I see:

```
C:\Users\Iain>s2aio

pymata_aio Version 2.28 Copyright (c) 2015-2018 Alan Yorinks All rights reserved.

Using COM Port:com17

Initializing Arduino - Please wait...
Arduino Firmware ID: 2.5 FirmataPlus.ino
Auto-discovery complete. Found 22 Digital Pins and 8 Analog Pins

The system cannot find the file c:\program.
```

[image]

MrYsLab, 2019-03-17:
Could you please try installing with a cmd window and not powershell. I suspect that poweshell may be causing the issue. Please let me know if that is successful or not.

IainColledge, 2019-03-17:
Uninstalled and installed using admin rights with a command shell, exactly the same.

MrYsLab, 2019-03-17:
I just installed using powershell, powershell with admin, and command with admin and in all cases I do not see the error you are. Here is a screenshot of powershell with admin:
[image]

When you installed Python did you check "Add python 3.7 to path"?
[image]

I also started the program by explicitly stating the path for the relevant files by using the -b option in conjunction with the -c no_client option. After starting the program, I opened up Scratch manually and and manually loaded an s2aio project.

C:\Users\Alan>s2aio -c no_client -b c:\Users\Alan\Appdata\Local\Programs\Python\Python37\Lib\site-packages\s2aio

IainColledge, 2019-03-17:
I've now installed on a laptop from fresh and works perfectly, looks like a local Python issue.

Thanks for your assistance, will close.

closed 2019-03-17 by IainColledge

jflpimentel, 2019-04-19:
I found same issue. There is a problem on Windows, if Python base path has spaces. That is the case when you install Python for all users: the path is "C:\Program Files\Python37".
The solution is to add a double colon before and after self.scratch_executable, on lines 216, 221 and 224 of file __main__.py:
[216] os_string = 'start /b "' + self.scratch_executable + '" ' + self.scratch_project
[221] os_string = 'nohup "' + self.scratch_executable + '" ' + self.scratch_project + ' ' + ' > /dev/null 2>&1 &'
[224] os_string = 'nohup "' + self.scratch_executable + '"> /dev/null 2>&1 &'

MrYsLab, 2019-04-20:
Thanks for taking the time to track this down. I tried applying your patches, and unfortunately, it resulted in a new set of issues after reinstalling Python for all users. I have updated the python install instructions on the wiki page to click on "Install Now" which installs Python for a single user. I can appreciate why you would want to install for all users, but at this point, I can only support install for a single user.

Thanks again.