MrYsLab/s2aio #27: Error "system cannot find the file..."

F-Coelho, 2019-11-07:
Hello MrYsLab
I' m using s2aio extension and it's fantastic. Thanks for the good work.

I' m installing on my computer but my user name is "Fernando Coelho". So when i try to launch s2aio command i get an error: 
The system cannot find the file c:\users\fernando. 

Can i correct this or should i have to create a neu user without blank space on the definition name?
Thanks in advance
[image]

MrYsLab, 2019-11-07:
Creating another user may solve the problem, but I think I have a solution so that you don't have to.
I am attaching a small python program, called base.py. 
I have to name it [base.txt](https://github.com/MrYsLab/s2aio/files/3820381/base.txt)
so that I can paste it in this message. Click on the link to download and then copy the code and place it in a file called base.py. Next run base.py as follows:
```
python base.py
```
Here is the output I see on my computer and you should see something similar for your user name.
```
C:\Users\Alan\AppData\Local\Programs\Python\Python37\lib\site-packages/s2aio
```

if you then start s2aio with the -b option, placing that string in quotes, that should start the program.

```
s2aio -b "C:\Users\Alan\AppData\Local\Programs\Python\Python37\lib\site-packages/s2aio"
```
Please let me know if that solves your problem. If it does, you may be able to create a .bat file to start the program up for you without all that typing.

MrYsLab, 2019-11-07:
Could you please type the following command:

```
s2aio -v
```

This should show the version of s2aio you are using and a path to the s2aio files.

Here is what my output looks like:
```
C:\Users\Alan>s2aio -v

s2aio version 1.17 - 1 March 2019
Python path = c:\users\alan\appdata\local\programs\python\python37\lib\site-packages/s2aio

C:\Users\Alan>
```

Does the path show your full user name?
If it does, then please continue reading. If not, could you please cut and paste the output and let me know what the full user name should be?
If you are using an older version of s2aio, please update by typing:

```
pip install s2aio --upgrade
```

Also, from the screenshot, I notice that pymata-aio is an older version. You can update by typing:
```
pip install pymata-aio --upgrade
```
Now to solve your problem. Using the path printed from s2aio -v, copy that path and place in quotes to start s2aio using -b option. Here is an example of what I mean:

```
s2aio -b  "c:\users\alan\appdata\local\programs\python\python37\lib\site-packages/s2aio"
```

Please let me know if that works for you.

F-Coelho, 2019-11-07:
Thank you very much for your answer.

But the result is the same.
I updated the pymata-aio.

[image]

MrYsLab, 2019-11-08:
I may have another possible solution. If this doesn't work, then I am afraid you will need to create another user name without a space in it.

Here is what I did:
1. I created a directory with space in it:
```
C:\Users\Alan\Documents\this\name it\s2aio
```

I then started s2aio with the following command:
```
C:\Users\Alan>s2aio -b C:\Users\Alan\Documents\this\name\\it\s2aio
```
It seems to work. If you notice instead of name it, I used name\\it. The space was replaced with a double backslash.

So in your case, replace fernando coelho with fernando\\coelho and try using the -b option again.

Please let me know if this solves the issue.

Thanks,
Alan

F-Coelho, 2019-11-08:
Hello again

I tried but same result.
No problem. I will work with anoteher account.
But thanks anyway from the help you provide.

Best regards
Fernando

closed 2019-11-08 by MrYsLab

HackEduca, 2019-11-09:
I just sent you a pull request to fix this issue.
Hope it works to Fernando.

Best Regards
Edson Sobreira
HackEduca

MrYsLab, 2019-11-09:
@F-Coelho 
Version 1.18 that addresses this issue is now available on pypi.
Thanks @HackEduca

F-Coelho, 2019-11-09:
Thanks you very much guys

Problem solved.