MrYsLab/s2aio #31: Blocks are not added to Snap!

kleag, 2022-09-04:
Hi,
I followed the instructions to run s2aio with Snap! (scratch seems to be more and more difficult to run on Linux unfortunately). When I start s2aio, Snap! opens but without the blocks.

Here is the output:
```
pymata_aio Version 2.34 Copyright (c) 2015-2018 Alan Yorinks All rights reserved.

Using COM Port:/dev/ttyACM0

Initializing Arduino - Please wait... 
Arduino Firmware ID: 2.5 FirmataPlus.ino
Auto-discovery complete. Found 20 Digital Pins and 6 Analog Pins
```

The board is an Elegoo Uno R3. I tried also with an old Arduino Uno R3 smd.

MrYsLab, 2022-09-04:
@kleag Unfortunately, the instructions are somewhat incomplete. I can explain how to load the blocks, but there is a more significant issue. Both Firefox and Chrome do not allow mixed content. The s2aio code uses HTTP, and Snap! is HTTPS enabled. 

It has been quite a while since I used Snap! so I don't recall if it was always HTTPS or if older versions of the browsers did not check for mixed content.

In any case, s2aio is now incompatible with Snap!.

To add the blocks to Snap!, select the file for the language you wish to use, and download the file.
In Snap!, click the file icon in the upper left, and select Import...

[image]

Most of the blocks will appear in the Sensing section:

[image]

Making s2aio https compliant is a complex issue that I do not have the time to address. However, if you feel you would like to try to solve it on your own, [this link may be a good starting point](https://stackoverflow.com/questions/51645324/how-to-setup-a-aiohttp-https-server-and-client).

Also, asyncio has changed and [this line ](https://github.com/MrYsLab/s2aio/blob/9947cde92ccfce851b5a2184210988b18ea125da/s2aio/__main__.py#L911) would need to be changed to these two lines:

```
loop = asyncio.new_event_loop()
asyncio.set_event_loop(loop)
```

If you have any questions, I would be happy to answer them.

MrYsLab, 2022-09-05:
I didn't think of that solution, which is an excellent one! I will leave this issue open, but since Scratch 2.0 is long gone, I will update the code and repository to be specifically for Snap!. I will generate new documentation explaining how to install the blocks and how to download a local copy of Snap!. 

I will have to perform additional testing because the pip command has changed over time. When s2aio was written, it was necessary to use sudo with Linux. The new version places the files in different locations than when pip used sudo. I need to test for Windows as well.

It may take a while for me to update s2aio, but since you could find a workaround, the pressure is lowered to give this immediate attention.

Once again, thanks.

kleag, 2022-09-05:
Great! Thanks again a lot for your work.

I ran pip install from a user virtual environment.

Note that I did not have to change the line concerning the event loop for the program to work. I tried with and without the change and saw no difference.

Maybe you could integrate a snap.html copy in your repository? I did not check the licenses, though.

MrYsLab, 2022-09-05:
Another good idea. Snap! uses the same license as I do, so there should not be any licensing issues. I will also look to add FirmaPlus to the Arduino library manager. It will make things much simpler to install.

I will make the event loop change, but I am glad it did not cause any issues for you.

kleag, 2022-09-05:
Wow, that will be great! I have an Arduino MKR1000, so with a WiFi chip too. Will it be usable also?

MrYsLab, 2022-09-05:
I may have been overly optimistic. I will probably need to create a separate version of s2aio for some of the boards. I am not familiar with MKR1000, but I may be able to reuse the ESP8266 server using a few #ifdefs. I can easily determine if the sketch compiles for the MKR1000, but would rely upon you to test to make sure that it works.

MrYsLab, 2022-09-08:
@kleag I found [snap4Arduino](https://snap4arduino.rocks/), and perhaps it can provide all that you require. This is not my code, but since the author works closely with the Snap folks, it is probably of good quality. 

Please let me know if it meets your needs or if you still wish me to pursue enhancing s2aio. 
Thanks.

MrYsLab, 2022-09-10:
@kleag I am closing this issue. The snap4Arduino site appears to satisfy your needs, including WIFI support already. I can't see the benefit of creating a competing solution. 

If you disagree, please let me know, and I can reopen this issue.

Thanks for your suggestions and solution.

closed 2022-09-10 by MrYsLab